wernerdaehn/CC3D-CableCam-Controller #9: Problem detecting channels

cofhal, 2018-02-18:
With the latest firmware verisons, I have a problem, I use a futaba receiver, but when checking the channel entries with the $ i command, it only detects channels 1,2,3, 5, but for channels 5,6, 7,8 and 9 the values always appear in 0, with the oldest versions if it detects these channels well.

Old Versions
$i
$i 1  6  7  5  8  256
last valid RC signal for channel  1 = 1024  (used as speed signal input)
last valid RC signal for channel  2 = 1018
last valid RC signal for channel  3 = 1010
last valid RC signal for channel  4 = 1020
last valid RC signal for channel  5 = 1909  (used as max acceleration value selector)
last valid RC signal for channel  6 = 1909  (used as programming switch)
last valid RC signal for channel  7 = 139  (used as endpoint switch)
last valid RC signal for channel  8 = 1909  (used as max speed selector)
last valid RC signal for channel  9 = 144
last valid RC signal for channel  10 = 144
last valid RC signal for channel  11 = 1025
last valid RC signal for channel  12 = 1025
last valid RC signal for channel  13 = 1025
last valid RC signal for channel  14 = 1025
last valid RC signal for channel  15 = 1025
last valid RC signal for channel  16 = 1025
current ESC out signal Servo 1 =  1500
 OK*4d

Last Versions
$i 1 6 7 5 8 256
last valid RC signal for channel  1= 1024 (used as speed signal input)
last valid RC signal for channel  2= 1016
last valid RC signal for channel  3= 1010
last valid RC signal for channel  4= 1020
last valid RC signal for channel  5= 0 (used as max acceleration value selector)
last valid RC signal for channel  6= 0 (used as programming switch)
last valid RC signal for channel  7= 0 (used as endpoint switch)
last valid RC signal for channel  8= 0 (used as max speed selector)
last valid RC signal for channel  9= 0
last valid RC signal for channel  10= 0
last valid RC signal for channel  11= 1025
last valid RC signal for channel  12= 1025
last valid RC signal for channel  13= 1025
last valid RC signal for channel  14= 1025
last valid RC signal for channel  15= 1025
last valid RC signal for channel  16= 1025
current ESC out signal Servo 1 =  1500

wernerdaehn, 2018-02-18:
Sorry, obviously was overprotective at some point. Will look into it tomorrow, okay?

cofhal, 2018-02-18:
I solved it, changing the end points of those channels to 80%, normally those channels were at 140%, but my radio a futaba T10CHG, does not allow to change the endpoint thel channel 9 or 10,

wernerdaehn, 2018-02-19:
The code there is using assumptions about the length of the high/low signals in order to understand if that is a SumPPM signal, a servo signal or something completely different like a serial signal of a SBus receiver due to a wrong configuration.
I know that there are similar but different timings possible with the various SumPPM receiver, e.g. when using 32 channels. And obviously when pushing the sender to a wider servo range.
If you happen to be able to record one entire SumPPM packet with the actual timings, say via a logic analyzer or so, that would be appreciated. I would then try to relax the timing constraints in the code.

closed 2018-04-02 by wernerdaehn