wernerdaehn/CC3D-CableCam-Controller #10: Just burned a motor controller

mitarashki, 2018-02-20:
Hello again Werner
I have just burned a RoboClaw 2x60AHV, 60VDC Motor Controller
Link: http://www.ionmc.com/RoboClaw-2x60AHV-60VDC-Motor-Controller_p_12.html
Dtasheet: http://downloads.ionmc.com/docs/roboclaw_datasheet_2x60AHV.pdf

I know it was not a problem that came from the CC3D board or software and I am just looking for a help to figure out what happened. 
Everything was running fine on a 6S (25.2V) LiPO battery until I have decided that it is ready to be tested on a 12S (50.4V) LiPO. Once I plugged it to the power it burned into flashes and smoke (ouch). This model is rated up to 60V,60A per channel and I ran it in "Bridged" mode which doubles the output to the motor to 60V,120A. 

I am just looking for your opinion what may be the cause of the burn. I am sure it was wired correct, but not sure that I have it properly tuned. 
Please excuse me for bothering you with my problems.

Thank you very much in advance for your support

wernerdaehn, 2018-02-20:
Can we discuss that somewhere else?
Maybe here?
https://www.rcgroups.com/forums/showthread.php?2103895-Cable-Cam/page9

mitarashki, 2018-02-20:
Yes sure

closed 2018-02-20 by wernerdaehn