wernerdaehn/CC3D-CableCam-Controller #11: Bluetooth connection

Willyfan, 2018-02-27:
HI Warner, I tried to connect a bluetooth module, it is not claer as setup it. I seen a images where you also changed the hall sensor IN. Can you help me?

wernerdaehn, 2018-02-28:
What board do you have precisely? The white one like I used originally?
It has the servo ports at one side - and, I have not changed anything there - and some more pins on the other edge. The top ones called TX3/RX3. These are the bluetooth pins.
Next step is to enter the command $B to autoconfigure the bluetooth module.

Willyfan, 2018-02-28:
Hi Werner, thanks for your answer.I have the new board, the same you have in the image directory with indicated Bluetooth pin. About Bluetooth, i solved because I found the command for configuration. It's OK ant it work.But I seen in the new image that you moved the input of one of hall sensor to the other side of the board. I will send you the pics later, now I'm on phone.

wernerdaehn, 2018-02-28:
I have not so much moved the Pin of the hall sensor, it is rather that this Servo6 Pin is actually not Servo6 on the board.
To be more precise: PA0 and PA1 are Servo5&6 on the white PCB but routed differently on the Flip32.
https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/_images/FLIP32-F4.jpg

kornel508, 2018-03-07:
Dear Werner!

I would like to upgrade my FPV (white PCB) with bluetooth module. Succesfully wiring the panel, but in the terminal after $B command i get an error. No response.....
I try the module to program with AT commands from an Arduino Uno. It's working good.
But from the FPV isn't working. What could be the problem? Thanks for your help!
[image]
[image]

Willyfan, 2018-03-07:
Are you sure that the module is powered? In pictures I don't see the led light... But as is blinking, maybe the picture took the off state.

wernerdaehn, 2018-03-07:
I would have the same immediate idea. Most of the pins supplying +5V or Vcc on the CC3D Revo are not powered by USB. So when you try to use the Bluetooth module, does the controller get power from e.g. the ESC? Or is it USB only?
Easy to find out, the HC-05 has a status LED that blinks while nothing is connected. Does it blink?

kornel508, 2018-03-07:
Thanks for your fast answer! Of course, the power LED blinks. 
There is a button on the upper left corner of the PCB, this is the programming mode switch. I hold this button when i powering the boards to step into the programming mode...
All function working from an Arduino. Please write how to program the Bluetooth module? 
AT+ORGL
AT+NAME=CABLECAM
AT+PSWD=1234
AT+UART=38400,0,0

Is this correct for the communication?
Thanks!

Kornel

wernerdaehn, 2018-03-08:
AT+BAUD6
AT+NAMECableCam
AT+PIN1234
refer to [bluetooth.c](https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/src/bluetooth.c)

Seems you do not have a HC-05?
Anyway, if you configure the BT module by hand and connect it, it should work right away. 38000,n,1 is the UART3 config.

```
static void MX_USART3_UART_Init(void)
{
    huart3.Instance = USART3;
    huart3.Init.BaudRate = 38400;
    huart3.Init.WordLength = UART_WORDLENGTH_8B;
    huart3.Init.StopBits = UART_STOPBITS_1;
    huart3.Init.Parity = UART_PARITY_NONE;
    huart3.Init.Mode = UART_MODE_TX_RX;
    huart3.Init.HwFlowCtl = UART_HWCONTROL_NONE;
    huart3.Init.OverSampling = UART_OVERSAMPLING_16;
    if (HAL_UART_Init(&huart3) != HAL_OK)
    {
        _Error_Handler(__FILE__, __LINE__);
    }
}
```

kornel508, 2018-03-08:
I have lot of HC-05, but your commands is for HC-06, this is a difference in AT comannds and HC06, it does not wait for any termination character for each AT command entry, HC-05 wait NL & CR after command.
HC-05 is an universal type for master and slave mode. HC-06 is a fixed slave mode device...
http://www.instructables.com/id/AT-command-mode-of-HC-05-Bluetooth-module/

wernerdaehn, 2018-03-08:
Okay, thanks for the correction. Given that the behavior is completely different, you have to press a button, commands are different, <CR><LF>, etc.... what is your suggestion?
I will obviously update the documentation but do you want to see more than that? Or are you okay configuring the HC-05 externally and just connect it?

Sorry for the confusion.

Willyfan, 2018-03-08:
> Dear Werner! No problem, I program manually the HC-05 with the good commands from your comments and working good. Thanks for your help!
> At weekends I would like to make a simple GUI application, for easy programming and position checking.
> Best regards
> Kornel

I wrote a app, working (only android) for checking and programming by Bluetooth, please let me k ow if you want try it.

kornel508, 2018-03-09:
Of course I would like to try it. But, IOS version would be more pleased. :) I know the HC-05 not work with Apple phones...
Thanks!

wernerdaehn, 2018-03-09:
Me too, but Android is fine. If you need more/different commands in the CableCam controller, let me know.

cofhal, 2018-03-09:
Dear Willyfan
Is possible public the APK for programming the CCD3.

Willyfan, 2018-03-09:
Yes, I'll upload the apk in the night on my server, and give you a link. Sorry for the late in reoply, but I was out office for production.
The app, I'm sorry, is only for android. And plaese, test it and give me a feedback, is a "beta" but it seems workwell.

Willyfan, 2018-03-09:
http://william-fanelli.com/wp-content/uploads/myupload/ID_CablecamSetup_5.apk

Please check it, let me know. Foe now, I don't need different command, but a OFFSET for limits can be useful.

Willyfan, 2018-03-11:
I re-uploaded now the file, because I found a bug in channel assignement. If someone just downloaded, please re-download (last upload at 11/03/2018 16:00  )

wernerdaehn, 2018-04-02:
Can you please check your app with the firmware I just uploaded? I made a couple of minor changes in the protocol and have the fear, depending on the logic you use, the app might not parse it correctly and needs slight tweaking hence.

Willyfan, 2018-04-02:
I will test it tomorrow in the afternoon. But I took a look at the your updated commands table, In my opinion the app work but I will test. I don't use the $j command, but all the old command should be work.

wernerdaehn, 2018-04-02:
It is things like trailing blanks got removed from decimal numbers, e.g. 
before '   0.00' and now '0.00'.
Or in $i I changed it that the OK comes before the channel assignments are printed in detail. Before all text was part of the command return. Now it is just the $i line with the channel list and the details are printed as extra text.

Willyfan, 2018-04-03:
Hi Werner, the app still work. Please explain me how work the automatic mode, so I can write additiobnal command to the app. Beside, a command for read fw version will be useful in order to mantain backward compatibility.

wernerdaehn, 2018-04-03:
https://github.com/wernerdaehn/CC3D-CableCam-Controller/issues/16#issuecomment-378088628

closed 2018-05-13 by wernerdaehn