wernerdaehn/CC3D-CableCam-Controller #12: Wrong argument number for ESC_NEUTRAL

Willyfan, 2018-03-03:
Dear Werner, ther is an error about argument number in PROTOCOL_ESC_NEUTRAL
If you send only 2 argument, you receive error from data parser, 
**$ERROR: invalid value for provided argument(s)**

 if you send 3 argument you receive error for argument number
**$ERROR: wrong number of arguments**

So it's impossible to set this values. 
I think that you added the third argument for max deflection but you don't changed the argument number check.

regards

 

```
 case PROTOCOL_ESC_NEUTRAL:
    {
         int16_t p[3];
         argument_index = sscanf(&commandlinebuffer[2], "%hd %hd %hd", &p[0], &p[1], &p[2]);
         if (argument_index == 2)
         {
             if (p[0] > 500 && p[0] < 2000 &&
                     p[1] > 0 && p[1] < 100 && p[2] > 300 && p[2] < 900)
             {
     
```

wernerdaehn, 2018-03-03:
True. But I can change it on Wednesday, only. Gone Skiing.

wernerdaehn, 2018-03-07:
Corrected with latest commit.

closed 2018-03-07 by wernerdaehn