wernerdaehn/CC3D-CableCam-Controller #13: flashing the firmware on CC3D EVO

TillMack, 2018-03-21:
Hi Werner,
I have an issue flashing the firmware on my CC3D EVO board.
Connecting the board, while keeping the boot button pressed ( unfortunately the board doesn't have a boot button, but a solder-pad-bridge), starts the internal USB bootloader. Shown by the Power LED, which is on and the Status LED, which is off.
But for some reason the board doesn't appear in the DfuSe utility at all.
I've done this procedure a couple of times, but the board never showed up in the DfuSe utility. 
Do you have an idea how to solve this problem?
Or know what I am doing wrong?

Regards,
Till

wernerdaehn, 2018-03-21:
I fear the CC3D EVO is a CC3D (STM32F1 based) and not a CC3D REVO (STM32F4 based).
https://www.reddit.com/r/Multicopter/comments/3f5r3f/cc3d_evo/ 

No surprise that it does not show up in the DfuSe tool. The STM32F1 does not have a bootloader wired in the MCU.

TillMack, 2018-03-21:
All right, didn't know there is a difference between these two boards regarding the STM32. 

Makes totally sense.
So I'll better get a CC3D REVO.

Thank you for your fast reply.

closed 2018-03-21 by wernerdaehn