wernerdaehn/CC3D-CableCam-Controller #14: broadcast to the Servo2 output

zond-media, 2018-03-21:
Thank you for this wonderful project!
Can you add the ability to broadcast directly 1 channel (any free) from the transmitter to the second output of the board (Servo2)? That would be great!
Plan to put the Servo to control Yaw on gimbal.

wernerdaehn, 2018-03-21:
That's actually a good idea. I have a similar request for a break-servo and was wondering how to incorporate that. Answer is simple: Use the RC sender mixer.
Will do.

zond-media, 2018-03-21:
in the case of a Brake use a Mixer is of course a good idea, but in my case it does not fit at all.
Thank you, looking forward to the implementation ;)

zond-media, 2018-03-22:
as another proposal ;)
It would be cool to have an automatic mode so as not to touch the throttle stick!

wernerdaehn, 2018-03-22:
btw, what gimbal are you using? Just out of curiosity...

wernerdaehn, 2018-03-25:
I just went through the hardware layout and the code, servo2 actually is configured for PPM servo output already, just like servo1 for the PPM ESC output. Hence to enable what you requested is essentially one line of code or so.

My problem is the $i command as such. Currently it takes as parameters

- ESC channel
- programming switch channel
- endpoint button channel
- max acceleration dial channel
- max speed dial channel

And now I add the aux channel, later the yaw, pitch, tilt channels for the gimbal and more. I think I want to change that and rather move the accel and speed channel settings to the corresponding commands $a and $v. Likewise with the aux and gimbal channel settings. Move those to commands of their own.

Agree? Leave $i as is but move he future gimbal channels to something else? And Aux to $i or to the gimbal commands?

I am undecided what's best.

zond-media, 2018-03-25:
I think that the best option to place commands for gimbal or aux into a separate group. for example $G
I'm sure this will be the right decision.

wernerdaehn, 2018-03-25:
Agree. Nevertheless, because the Aux Output can be used for Yaw and Brakes, I thought I better add that as yet another $i parameter. That does not prevent us to assign the value via other commands as well or do like suggested for the full gimbal control.

Regarding Gimbal Control I am preparing everything to use another Pin as SBus out. Unfortunately the Revo board has a single SBus Inverter only (none on the TX pin), hence we need another piece of hardware for that. Too bad.

zond-media, 2018-03-25:
**wernerdaehn**, thanks!
and what about Automatic Mode? ;)
Can I open new Issue?

wernerdaehn, 2018-03-25:
Can you elaborate on what you mean with automatic mode? This is related to the break as opposed to the initial inquiry of yaw control, isn't it?

zond-media, 2018-03-25:
I mean automatic CableCam movement at a given speed back and forth without moving the gas stick with the start and end position. This mode is needed for working alone, so that there is a possibility of controlling the gimbal.

wernerdaehn, 2018-03-26:
Yes, a new issue please.

zond-media, 2018-03-26:
Ok

closed 2018-03-26 by zond-media