wernerdaehn/CC3D-CableCam-Controller #15: Speed and acc.

Willyfan, 2018-03-23:
Hi, now I'm trying for the first time the controller, and with my cablecam I have a problem with speed and acceleration.
In passtrough mode is ok, but I don't have limits.
In passtrough with endpoints, the response of the motor is not good. When I put the stick in one direction, it accelerate very slow (I tried many setting for this) and when I release the stick, it decelerate very slow, too slow.
If I rotate at max value the pot for max acc and max speed the situation is better, but not good.
Question is: is possible, when pot is at max value, have the same response as passtrough? 
The problem is that on short cable (under 15 metres), I'm unable to reach good speed before end point. 
Regards
William

wernerdaehn, 2018-03-24:
I have made some changes last month and the acceleration - using the poti on the RC sender - so that min acceleration is 20 secs from 0 to 100% and the maximum 2 secs.

Now we need to find out what it is. Is it that everything works fine but the 2 secs is a too little max value?
Is it the poti range?

Can you set the potis to max speed and max accel and give me the readings for $a and $v?

Keep in mind, if you change your input so that the potentiometers are not used, you can set any value for $a. Might be worth a test.

Willyfan, 2018-03-24:
Hi Werner, thank you for your reply .
I done some test. If I disable POT, I'm able to set all value from 0 to 0.999 for acceleration, in Programming and Operational mode.
With POT, after setting at best the max deflection, I'm able to have value from 0 to 0.500 (no more) and pot work only in operational mode (reading the value from putty with $a command). Programming mode say always as set by $a command.
With this value the acceleration and the deceleration are too slow in some cases. Also the value set to 1 with POT disabled have a deceleration too slow in some cases, but it can be accettable.
So, I think that a modification of the code can be extend the POT value range from 0 to 1 and not from 0 to 0,500. Or, if is possible, the POT set the value from 0 to max value setting ($a). In this way, You can limit anyway the MAX acceleration and deceleration, but if you need you can have a wider range.
I think that the same can be applied to speed, but I must do some other test for speed.

wernerdaehn, 2018-03-25:
No problem doing that. But before I do that actually, let me ask a few questions.
A setting of 0.5 means 2 second ramp from 0 to 100% speed. A setting of 1 means one second. Isn't that a bit aggressive? 
Assuming it is all fine, why can your cablecam accelerate so fast? Do you have such a low max speed or better components?
What ESC are you using? VESC via UART or a regular ESC via Servo1 (PPM output?)

wernerdaehn, 2018-03-25:
Okay, new firmware allows to change $a from 0 .05 to 1.00 via the RC.

Willyfan, 2018-03-27:
Thank you, now I will try. Anyway, I noticed that my problem is not the acceleration (also 2 seconds and more can be good) but sometimes the deceleration, because when you release the stick the cablecam move for other 2-3 or more seconds. Sometimes is too much.

wernerdaehn, 2018-03-27:
Okay. Makes my question regarding "What ESC are you using" even more important.
Take the VESC for example, as this has a PID closed loop for speed. Even here it does not help much if I tell the VESC to reduce the RPMs to zero in 2 seconds, if the physics or the PID settings do not allow for that. Imagine the cablecam 50kg heavy and a 1kW motor. No way it can break that quickly. 

And the cablecam controller has no way of knowing that. The only thing I can do is a failsafe: When the breaking is not powerful enough and the end point is overshot significantly, forget about the ramp and brake as hard as the ESC can. In case of the VESC that means setting the speed to zero, in case of the PPM output setting the speed signal to neutral.

Willyfan, 2018-03-27:
You are right: for now, at the question "what ESC" my answer is a XERUN xr8 (tried sensored and not sensored, little difference). I want try a VESC that I have at home, but for now is a bit experimental. I hope that I will find time after 15 april for this test on VESC.
Now, I updated my board with new firmware, I will test on cable tomorrow morning.

Willyfan, 2018-03-27:
Dear Werner, i noticed that beside to acc modifications you added 2 parameters to input channel mapping. I read that one is for AUX, and the other? I will modify the android app for this. Have you tried it? Anyway, old version still work, but ignore thet parameter.

wernerdaehn, 2018-03-27:
The last one is the channel for the Aux output. The one before last is the channel to switch between Passthrough, Limiter, Limiter+Endpoints via the RC.

No, haven't tried the app yet.

Willyfan, 2018-03-27:
Ok, I will add also the last parameters as soon as possible.
I can use a 3 way switch for the last switch? What values it need from RC?

wernerdaehn, 2018-03-27:
Exactly, a three way switch with high, neutral, low as signal sets the three modes.
(When USB or BT is connected, flipping the switch writes an event text. You can use that to validate.)

Willyfan, 2018-03-27:
Ok, It work.  I uploaded a new version of android APP, same link as before.
I added the new parameters, and a continuos reading of message from cablecam, so you can read swich alert. About acceleration, now is better controlled. Anyway, tomorrow morning I will test on cable.

Willyfan, 2018-03-28:
Tested on cable this morning, I'm very happy. Now response of the cablecam is very good also at low speed. Acceleration have no problem at max, but I checked also with pot at mid position, is ok.
When speed is hight, the cam overshot the limits: need to be a bit "conservative" when setting the limits and use high speed.

Willyfan, 2018-03-28:
I agree, this is a good setting.  Maybe better precision can be obtained with a  logaritmic curve, but I can set it on the transmitter, no problem.

wernerdaehn, 2018-04-04:
Got a request from somebody else to change it yet to another value. As can't do it right for anybody, I revised the logic.
With $a 0.500 0.050 you set the absolut max values in Operational and EndpointProgramming mode: 1/0.5=2 seconds from 0...100% and in safemode 1/0.05=20 seconds from 0...100%.
With the poti then allows you to scale the actual max accel between 0 (almost zero) and this programmed max.

Speed is the same thing, use $v 1.000 0.250 to limit the speed stick movement and the poti then scales it further down. Here the logic makes a little less sense, why would you set the speed limiter in normal mode to anything other than +100%??? And the minimum speed is certainly not almost zero but 20%.

wernerdaehn, 2018-04-22:
After another chat, I changed the speed logic again. Before a speed limit of 50% meant every speed movement >50% was unused.
Now the stick can be moved unlimited but the speed limiter is applied to the output.
Example: 
Servo 1 output: Max speed of 50% means the the output signal moves between 1500+-400 instead of 1500+-800 (or whatever the precise numbers are in $N).
VESC Uart Output: Max speed is 50% of the max erpm value configured ($e).

As a result the stick can be used at full range to control the speed between +- max speed.

wernerdaehn, 2018-05-13:
I am closing this as DONE.

closed 2018-05-13 by wernerdaehn