wernerdaehn/CC3D-CableCam-Controller #19: How accurate is the position reading?

wernerdaehn, 2018-05-21:
Can you please keep an eye on the position sensor?
In my case I marked one magnet of the sensor wheel with a pen and hence know the exact position. And I marked the rope at position 0.
When I drive back and forth, high & low speeds, high & low accelerations, the point does not move at all. Always use the $p command to get the position. Just saying it does not stop at the end point precisely has too many other variables. Mass of the cablecam, brake momentum etc.

I had to tweak the STM encoder settings a bit but now I believe it works perfectly for me. No slippage as the sensor wheel can move freely and is pressed via the rope tension into the rope. And no hall signals get lost.

There could be a couple of reasons why you results might be less accurate and I want to get a feel how well all works at the moment.
- Magnets induce a magnetic field in the alloy causing the hall sensors to trigger.
- The distance between the hall sensors in relation to the magnets is not optimal. I have 8mm between magnets, 13mm between the two hall sensors. Keep in mind, there is a hysteresis in the hall sensors and the direction is derived from the fact which flank fires first - sensor1 or sensor2. Currently I believe 8mm/8mm would have been better given that we alternate the magnets north/south and the hall sensor fires on north only.
- Capacitors near the hall sensors needed or not? I do not use any at the moment.

zond-media, 2018-05-22:
@wernerdaehn, I tested the latest firmware version. works terribly. does not correctly record end points after battery replacement. flies the endpoints even further than the previous firmware.
return it as it was, please ;)

wernerdaehn, 2018-05-22:
It should not be more than 44 steps more than before (I sensor wheel revolution). Is it about that distance or way more?

What do you mean with "does not correctly record end points after battery replacement"? 
And for $r, you have programmed it or just read it? What is the value before you set the endpoints? (Because if $r=0 then the end point programming tries to guess it)

wernerdaehn, 2018-05-22:
I have made the changes but not tested them. Not sure when I can test but feel free to give it a try.

wernerdaehn, 2018-05-22:
If you just read $r, then please program it as well. $r 1 and $w.
In the firmware of today I test if $r = 0 and program it together with the end points.

zond-media, 2018-05-22:
Thanks! I will test in the first half of the day tomorrow and write the result. about the command $r 1 understood, I will do as you write

zond-media, 2018-05-23:
Need ftdi module support like this [http://rctimer.com/?product-834.html](url) on rx3-tx3 pin. Can you do that?

wernerdaehn, 2018-05-23:
Simple UART, so why not. 
The original CC3D Revo has a 433MHz module onboard as well. It is connected via SPI and I did like Bluetooth better (because then I don't need a receiver for the tablet). But I can enable that as well, if you like. Just a thought.

zond-media, 2018-05-23:
Perhaps the best solution would be to buy this module  http://rctimer.com/?product-834.html ?

wernerdaehn, 2018-05-23:
Yes, it is cheaper and works out of the box. Only thing I don't like is that it is yet another PCB just like today I use the bluetooth module. With the original CC3D Revo the 433 Module would be onboard already.
I just looked for a CC3D with onboard OPLink and did not find any in Europe, though.

zond-media, 2018-05-23:
I’m just now upload new firmware. Nothing has changed...

wernerdaehn, 2018-05-23:
Video please. Showing where you program the start point, how it brakes there, what happens when you run the play command, what happens to the start point when you restart the cablecam. Okay?

zond-media, 2018-05-23:
Strangely, now the position is saved and loaded correctly, but the move for the end points is still great. In the photo I showed the installation of the end points at a distance of about 6 meters. https://drive.google.com/file/d/1n0BcQoQk7_kxnT1lFSauiGHxpjp1nHZ0/view?usp=sharing
https://drive.google.com/file/d/1EBWfAJsX7KWBaN2xs0utcCGs9_Ifmju2/view?usp=sharing
https://drive.google.com/file/d/1Ek4Lk84NlM8VU_OAN6_rcWTKZwFNlvDa/view?usp=sharing

wernerdaehn, 2018-05-23:
I think I know what is going on. It is not so much related to the firmware of last weekend but a few days before that.

Please set $g 10 and save it. That causes the ESC to brake more aggressively and not just once the endpoint has been overshot. 
NOTE: $g 0 disables it, so that's no good.

The controller can output 2 kinds of values via PPM: Slowly reduce the thrust or neutral.
When you listen to the motor, it makes a different noise when passing the endpoint. It brakes heavily once the end point has been overshot but none of that noise is heard before reaching the endpoint. It rather looks like free wheeling while getting closer and closer to the end point.

And that makes sense:
- When the brake distance based on the current speed is calculated to be slightly over the end point, the controller does reduce the speed signal. In your case it does slow down but not enough to stop near the end point.
- When the cablecam passed the endpoint, a failsafe is triggered. Now I don't care about the speed but set the output to neutral. For your ESC that is the signal to brake as hard as possible with the drag brake.
- When the brake distance calculation already shows the endpoint will be overshot significantly, I put the output to neutral as well. That is the emergency brake. Idea is, if the speed = 100 and the brake distance is 5000 steps beyond the end point, it is better to brake as hard as possible now already.

The change affecting you is the last one. The emergency brake $g distance defaults to 100. Which is quite a lot in absolute terms. But on the other hand, at very high speeds, the brake distance might easily be 100 steps higher for a few milliseconds. The VESC does not like getting speed = 0 every few seconds and it is not necessary either. Therefore I have added a speed component into the emergency distance. At speeds close to 0 the emergency brake kicks in at 100 steps position. The higher the speed is, the higher the emergency brake distance gets.
With your controller an occasional neutral signal to engage the brake is actually desired.

    if (activesettings.max_position_error != 0.0f && (pos + brakedistance >= semipermanentsettings.pos_end + activesettings.max_position_error + 30.0f*brakedistance/activesettings.max_position_error))
    {
      controllerstatus.emergencybrake = true;

wernerdaehn, 2018-05-23:
Stupid me, the lower the value for the max_position_error is the smaller the last component in the if condition should be. Which is not the case. Let me change that quickly.....

wernerdaehn, 2018-05-23:
Updated Firmware uploaded.

zond-media, 2018-05-23:
Now, what I need! Thank you!
Add support for the FTDI module and the system will be fully ready for work!

wernerdaehn, 2018-05-23:
As said, the FTDI module will work already. Just need to configure it for 38400 Baud and connect it.

(Note: I will delete all these posts here in the next days as the actual intention of this github issue is to capture position sensor reading issues.)

zond-media, 2018-05-23:
How can I configure for 38400 ??

zond-media, 2018-05-23:
I'll try with PuTTY, but then I try to click on keyboard, no answer from controller

zond-media, 2018-05-23:
Ok, I'll configure to 38400 with 3DRRadio, but  in terminal I can't connect..

wernerdaehn, 2018-05-23:
Or the Air Radio Config tool as it says at the bottom of the product page you linked?
http://rctimer.com/?product-834.html

zond-media, 2018-05-23:
I did it. speed 38400, but nothing happens when I press keys in the terminal window

wernerdaehn, 2018-05-23:
Best would be to diagnose it without the cablecam controller in the picture. Connect the module working in the cablecam to you PC, with the 38400 baud rate and the ground station module as well. Since this is supposed to be a transparent link, every character you send in the one putty terminal should appear in the other and vice versa. 
Only when that works it makes sense to even look into the cablecam controller.

Agreed?

zond-media, 2018-05-23:
Because on Aux ch. I have digital servo with other neutral point

wernerdaehn, 2018-05-23:
??? A servo that has center not at 1500??? Never seen that. Unless somebody reprogrammed the servo - which is possible with some digital servos.
I would really like to avoid that. Do you insist?

zond-media, 2018-05-23:
I would be very very grateful if you do this!

wernerdaehn, 2018-05-23:
Done. See $A.

zond-media, 2018-05-23:
Thanks a lot! Tomorrow I’ll check it

zond-media, 2018-05-24:
Everything is working as I wanted, amazing accuracy at any speed. I was able to set the neutral point for the servo (= 1474). It is very cool! Thanks again for the excellent work! I hope for long-term cooperation!

closed 2018-05-24 by wernerdaehn