wernerdaehn/CC3D-CableCam-Controller #20: How accurate is the position reading?

wernerdaehn, 2018-05-24:
Can you please keep an eye on the position sensor?
In my case I marked one magnet of the sensor wheel with a pen and hence know the exact position. And I marked the rope at position 0.
When I drive back and forth, high & low speeds, high & low accelerations, the point does not move at all. Always use the $p command to get the position. Just saying it does not stop at the end point precisely has too many other variables. Mass of the cablecam, brake momentum etc.

I had to tweak the STM encoder settings a bit but now I believe it works perfectly for me. No slippage as the sensor wheel can move freely and is pressed via the rope tension into the rope. And no hall signals get lost.

There could be a couple of reasons why you results might be less accurate and I want to get a feel how well all works at the moment.
- Magnets induce a magnetic field in the alloy causing the hall sensors to trigger.
- The distance between the hall sensors in relation to the magnets is not optimal. I have 8mm between magnets, 13mm between the two hall sensors. Keep in mind, there is a hysteresis in the hall sensors and the direction is derived from the fact which flank fires first - sensor1 or sensor2. Currently I believe 8mm/8mm would have been better given that we alternate the magnets north/south and the hall sensor fires on north only.
- Capacitors near the hall sensors needed or not? I do not use any at the moment.

Willyfan, 2018-06-13:
Hi Werner, I'm waiting for a VESC in order to try also this issue. 
If you are interested, ther is a guy that sell clone of vesc 6, at around 130 pouds. You can fint him here: http://www.electric-skateboard.builders/t/hw6-4-based-esc-escape/37579/60

Willyfan, 2018-07-02:
I checked the precision of the sensor, and it is perfect. I have sometimes some problem because a little slip of the wheels when accelerate too quick. bu the sensor is ok, and beside mi wheels are in derlin. I will try putting some rubber o-ring on the wheel in order to eliminate the little slipping.
I'm using a old circuit for sensor, with 3.3 volts regulator and it work good. I have now some boards without voltage regulator, I will try it also.

DmitriySurin, 2018-07-27:
I spent two weeks to figure out the algorithm of the hall sensors step. Here it is: 1, 1+2, 2, 0. I missed 1+2 step in my configuration hence nothing works. Could you explaim more about 8/8 mm does mean in your guide above?

wernerdaehn, 2018-07-27:
Check the bottom of this page:
https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/Step%206%20-%20Connecting%20the%20Position%20Sensor.md 

The point is not that N/S poles are 8mm apart and so is the sensor, it should be 8mm to 13mm for example.
There are two effect:
1. When one sensor fires on rising/falling flank, the other sensor is read. This determines the direction. To make things more obvious, the goal is that first sensor1 gets a rising flank and then sensor2. Then sensor1 gets a falling flank, then sensor2. And in the reverse direction, it should be sensor2 first and sensor one after.
1. There is a hysteresis, so the sensor switches later once the north pole is in reach and switches later when the north pole gets out of reach. 

In other words, when turning the sensor wheel with a constant speed, there should be a pronounced time gab between leading sensor On and trailing sensor On. Is there?