wernerdaehn/CC3D-CableCam-Controller #21: Problem with mode switch

Willyfan, 2018-06-26:
Hi Werner,
I have to kind of problem with mode switch from RC
1) with old firmware, I remember that the switch work with minimum, mid and max value for switch with a quite wide range. On  my radio, now the 3 point are very closed. I can programming my radio in order to work, but I think that it is a bit dangerous.
For example, if my mid value is 990, at 984 mode switch to passthrough, at 1002 switch to passthrough  with limiter and at 1058 switch to passthrough with limiter & end points. 
A wider range is more safe.
2) If I'm  in programming mode i can change mode without problem, but if I switch to Operational mode after somequick changes sometimes the board don't responds anymore to RC command, but it work connected to the terminal. If you change mode slowly it work.
If you move the motor stick, or you change mode from terminal with $m command the board return to work and read RC again.
Last problem is very subtle, it happen sometimes.

wernerdaehn, 2018-06-27:
The second I did not get.

The first point I will change. At the moment only the neutral means a mid-position. I will increase that by a bit, e.g. neutral +- 30%?

Willyfan, 2018-06-27:
I will investigate more about the second point.
The first point, I understand, the mid point is only the neutral range, all under or over neutral switch.
I think that a 30% more can be safe.
So, if I have in mid (for example) 1000, I need to go under 700 or upper 1300 in order to switch. More strong.

Willyfan, 2018-06-27:
I check and problem was solved. I want understand the pint 2 of my post, anyway all is ok for now.
Beside, I done some upgrade to my app, and added a module for rope calculation.
Same link for who want try it:
http://www.william-fanelli.com/wp-content/uploads/myupload/ID_CablecamSetup_7.apk

wernerdaehn, 2018-07-13:
This can be closed, can't it?

closed 2018-07-13 by wernerdaehn