wernerdaehn/CC3D-CableCam-Controller #22: little bug in $i input parameters

Willyfan, 2018-06-27:
Dear Werner,
there is a little bug in $i input parameters. As now many people use $j instead maybe is not a big problem, but it is a problem for my app that set all channel input with one only command.
$i don't accept anymore 256 as value for disable the input. I tried to use it for many input  and if there is this value is skipped.

Willyfan, 2018-06-27:
I took a look to your code, the problem is related to SBUS_MAX_CHANNEL that is set to 16 in sbus.h
So, all value over 16 is skipped.
for $j command you use a fixed 256 as max value, so it work. 
If solve this bug is not safe, I can change the layout of my app in order to set channel in single  mode, please let me know.

wernerdaehn, 2018-06-27:
Just tell me what command you want to execute. I did not get the details or I am not sure I did.

Willyfan, 2018-06-27:
For example:
$i 1 2 3 4 5 6 7 8 Work and set all channels.
$i 1 2 3 4 5 6 256 256 (in order to DISABLE AUX and PLAY work only for the value 1 2 3 4 5 6 and NOT disable (value stay as before)
In my app  I need to send always all 8 parameters (and this work) but I can't disable channels if i enabled before with $$i command (it work with $j from terminal)

wernerdaehn, 2018-06-27:
Understood. Will correct that.

wernerdaehn, 2018-06-27:
done.

closed 2018-06-27 by wernerdaehn