wernerdaehn/CC3D-CableCam-Controller #23: Passthrough and speed limit

Willyfan, 2018-06-28:
Dear Werner,  yesterday night I noticed during some test on the board that the **passthrough mode** is influenced from max speed control pot, but not from acceleration pot. 
If I switch to **Passthrough with speed limits** all 2 pots works (and it is correct).
Also exponential factor change the response of Passthrough mode.
I remember that before this mode was input=output without filters.
It's a choice or it's an error? Also, is not a big problem as we can set max speed to 1 and take pot at max for test, but just to know.

wernerdaehn, 2018-06-28:
That would definitely be an error. If passthrough & operational (no endpoint programming) is set, then input=output.
The only calculation is done via the $n (input) $N (output) value ranges.

Willyfan, 2018-06-28:
About range is logical because you need to "translate" sbus in pwm.
But I'm sure that in passthrough & operational the speed pot work. I noticed because I forgot the pot at min, and motor start very slowly. Turning pot at max all work.
Acceleration pot don't have influence in this mode.

A question not related with this problem: I seen just some minutes ago in some your images that you have connected a yellow wire to CH6/TX6 pin on the board. What is? If I understand correctly, VESC is connected to Servo 3 & 4 - but this board have some different layout, so I ask...

wernerdaehn, 2018-06-28:
Okay, should be easy to check and reproduce for me. Keep you posted.

This is the Gimbal Output connected via an external SBus inverter to the Ronin. 
see last line in
https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/Implementation.md#hardware-mapping 

I plan to add individual pages for the hardware connections soon.

wernerdaehn, 2018-07-22:
Hmmm. Yes, you are obviously correct. In passthrough mode the speed poti does change the speed.
The question is, is that good or bad.

Purely from the implemented logic speaking, if there is an output range configured, say +-700us, then 100% forward should be a pulse 1500us+700us (neutral plus max).
The speed poti does change the max allowed value. So a poti of 50% reduces above example the max value to +-350us.

If you remember, initially we said the poti controls the stick movement. The result was that a speed limit of 10% did ignore all stick movements in the range of 10%-100%. Instead we changed it to:
The stick should use the full range and we scale the output. And that we do, we scale the output.

Changing that is simple, I just need to add another if-then-else when generating the output. If passhrough then scale based on the ESC output, when limiter is on then scale based on the speed limiter.

Question is, what is better? Currently I am for adding that. Other opinions?

Willyfan, 2018-07-24:
I think that the user expects that passtrough mode is a input output direct mode. In the real life, there is no reason for use passtrough mode, except when you test or check the system.
If you want only ignore the endpoints, passtrough with limits is perfect. 
So, in my opinion is better that poti don't work in this mode,  if is not a problem. We will see other opinion anyway.

closed 2018-09-22 by wernerdaehn