wernerdaehn/CC3D-CableCam-Controller #24: $P 1 for automatic play don't work

Willyfan, 2018-06-29:
I'm adding to my app the play function, but I discovered that the play mode work from RC but not from Bluetooth $P 1 command. I tried also to put $P 1 from Putty connected by usb, and it don't work.
The command is accepted as correct, but nothing happen.
Also tried to unassign the RC channel.
What I'm doing wrong?

Willyfan, 2018-07-01:
Again about Play function, I have some suggest:
1) very important: if you turn on the cablecam and forgot the play switch on in the RC, the cablecam start to play and it can be dangerous. I don't know how solve this, may be at the start up the swich must be in off, if is in on you must switch it in off before start operation.
2) the 5 seconds pause sometimes is too much. Can you add a parameter for set this time? Is not necessary that you can set from RC (RC have a lot of function now), simply from terminal (and my app)

wernerdaehn, 2018-07-22:
- Found the problem. I said "if the channel is unassigned or stick = neutral then Play = OFF". It must be "is unassigned then do nothing, else check if in neutral and turn on or off". 
- Yes, that is a very good point. We have a controller that prevent something, yet with Play and storing end points permanently, it might have its own will. I have changed the logic that it fall off the Play mode whenever there is even the smallest glitch. In all those cases you need to turn off Play in order to start it again.
    - Stick movement? Forcibly turn off Play.
    - Entered another mode? Forcibly turn off Play.
- The command $R 5 does set the the wait time when reversing to 5 seconds (default)

ATTENTION: I have also changed the command response for all commands. Before it was a mixture of OK and echoing back the settings plus OK.
Now it is consistent for all commands: Without arguments it does the same print as before. With arguments it sets the values and then does the same print as without commands.
Example: 
Old: $N 1500 30 700 ---> $N OK
New: $N 1500 30 700 ---> $N 1500 30 700 OK

Willyfan, 2018-07-24:
Thanks Werner, I will modify my app. Actually I'm in holiday until 13 August, so I can't do it now. I let you know when ready.

Willyfan, 2018-08-12:
During my holidays I had a problem at a my eye: retina detachment. So, I''m back before, from end of July.
My eye is OK now, and I' m working on cablecam. Now the command $P 1 is working.
I will publish link to new app asap, I'm checking it. Now the app require android 4.0 or higher.

closed 2018-09-22 by wernerdaehn