wernerdaehn/CC3D-CableCam-Controller #27: Bluetooth connection FLIP 32 and HC-06

DmitriySurin, 2018-07-22:
I have HC-06 manually programmed to:
AT
OK
AT+UART
+UART:38400,0,0
OK
AT+NAME
+NAME:cablecam
OK
AT+PSWD
+PIN:"1234"
OK
AT+VERSION
VERSION:3.0-20170609
OK

But it doesnt work for FLIP.

wernerdaehn, 2018-07-22:
That +UART:38400,0,0 worries me.
Can you change the settings to UART 38400,1,0? It should have no parity, one stop bit.

DmitriySurin, 2018-07-22:
No it doesnt help

wernerdaehn, 2018-07-28:
I have uploaded a firmware version with debugging support for the bluetooth UART.
- Connect the board to USB, open the terminal.
- Connect the bluetooth module.
- Via the USB terminal execute the command $d b. This turns on debugging. From now on, whenever the bluetooth module does send a char to the controller, this char is also printed in the USB terminal. In my example I send via my tablet the text abcd to the bluetooth module and on the USB terminal abcd gets printed.
- In the USB terminal execute $d B after you sent data via bluetooth. This command shows the lowest level error codes on UART3. Frame error, parity error, overflows, noise.

As said, this for monitoring the RX pin on the UART3 of the controller. 

I hope this helps us isolating the issue.

DmitriySurin, 2018-07-30:
I send 123456 via Serial bluetooth terminal and I can see very strange result in your debugging mode:

Bluetooth debug: ON
▒▒▒▒

DmitriySurin, 2018-07-30:
Also $d B said:

Bluetooth UART's last error code flag says: no errors

wernerdaehn, 2018-07-31:
That does not make sense. Sending different characters resulting in the same target character without frame or parity error is impossible. Something is wrong.
What I would do next is connect the PC via a USB-to-UART adapter to the cablecam controller's bluetooth pins (without 5V) and proof the UART is fine or not.

DmitriySurin, 2018-07-31:
What I would expect to see in terminal? At? Or what?

wernerdaehn, 2018-07-31:
If you send "123456" to the UART, in the USB terminal you would see "123456".

I have added a print statement into the UART3 receiver logic. Before the received char is put into the UART buffer, it is also sent to the USB terminal.

            void UARTX_IRQHandler(UART_HandleTypeDef *huart) {
                if (controllerstatus.bluetooth_passthrough) {
                    PrintSerial_char(byteReceived, EndPoint_USB);
                }
                huart->pRxBuffPtr[huart->RxXferCount % huart->RxXferSize] = byteReceived;
                huart->RxXferCount++;
            }

DmitriySurin, 2018-08-01:
Ok. I cheked it. Thats works. Flips UART is fine!

DmitriySurin, 2018-08-06:
I bought HC-05, and now everything is work fine.

closed 2018-09-11 by DmitriySurin