wernerdaehn/CC3D-CableCam-Controller #30: Deceleration problem

Willyfan, 2018-08-12:
During some test on the rope, I found a little problem with deceleration near to the end points. I see this in autoplay, but it is the same if you drive the cablecam to end points at max speed.
The problem is that the deceleration is not smooth, but a bit "choppy".
I solved the problem rising the max allowed error parameters from 100 (default) to 300.
Now the deceleration is smooth, but when the cablecam arrive to the end point speed is not zero (anyway is NEAR to zero) and it stop abruptly.
My mind is that the calculation for deceleration have some kind of error, so, if allowed error is low the board try to adjust speed creating the "choppy" deceleration. When max error is higher, The deceleration go until the cablecam don't is on the end point, but because this error speed is not zero, so the board brake in order to don't go over the end point.
My guess is that deceleration start too late, Or is less that we need. Maybe also cablecam inertia do something, anyway, is not possible to have a parameter for "anticipate" a bit the deceleration? Just for debug...

wernerdaehn, 2018-08-13:
We need to separate here between VESC and RC ESC. With the RC ESC there is not much I can do. I can put the stick slowly towards neutral - little brake force - or put it in idle - strong brake force. There is no _proportional brake_ in forward/reverse mode. Hence the best is a narrow max-pos-error of 100 to switch between both brake modes more often.

With the VESC we have conflicting requirements. We want to stop at the end point, we want not to drive over the end point and we don't know the inertia of the cablecam. (Also the inertia is not static - driving up/downhill, rope tension etc). I have tried multiple things like setting the end point 100 step early and drive at low speed to the very end. Nothing really did work well. 

Which of the two modes are we talking about and what is the suggestion? Also a video would be nice to agree it is the above or something that can be optimized.

Willyfan, 2018-08-13:
I'm using vesc and not rc esc.
I have a video on my phone. I'll publish it in the afternoon, but on video is not so clear what I mean.

DmitriySurin, 2018-08-13:
And I am using esc from skyrc. From my side I can say that long acceleration and decceleration affect more choppy brake. Like abs on the car. When I put low accell and deccel value my cablecam sleep on the rope at high speed while braking but there is no choppy effect discribed above. My suggestion to separate accel and deccel value for manual mode and for automatic braking near end points. Also it would be nice if we could independently adjust accel and deccel value.

DmitriySurin, 2018-08-13:
Hence the best is a narrow max-pos-error of 100 to switch between both brake modes more often. What does it mean? In my cablecam construction there is big gear ratio and I am using 4-6s battery. So from the esc side the drug brake is strong enaphf to hold the cablecam via big gear ratio.

Willyfan, 2018-08-13:
This is a video where you can see that the Cablecam brake an go during deceleration. As you can see, the Cablecam jump slightly near to the end point.
No problem with accelleration.
The video is with max error at default value 100.
Raising at 300 (or more) the deceleration is always smooth, but when the camera is at the endpoint speed is not zero, so it brake abruptly.
This is not a big problem, but if is possible reach speed zero without this effect (also if the camera went slightly over the endpoint, all is better. 

Anyway, is better the results with hight max error value than the choppy deceleration, for quality of video.

wernerdaehn, 2018-08-14:
What are the drag brake and initial brake settings on the ESC itself? Rather high values or low?

Imagine the following situation:
CableCam is 5m away from the endpoint and according to the accel settings it has to brake now. 
In the next cycle it is 4m away and the speed has not been reduced significantly. The brake distance calculation shows it will overshoot the end point by 1m. As you see, moving the stick slowly to neutral does not cause the cablecam to brake enough. What signal should I send to the ESC?

A few cycles later, it is exactly at the endpoint but has some excess speed left. It will overshoot the endpoint by 1m. What should I send to the ESC?

Willyfan, 2018-08-14:
> What are the drag brake and initial brake settings on the ESC itself? Rather high values or low?
I'm using a VESC6, so I don't set drag brake and initial brake.
Now, I tested the cablecam far from end point. If I move the stick at center, the cablecam  decelerate very smooth and stop (after I released the stick). But I guess that this is a combination from deceleration and camera inertia, and I can't know how much distance the camera stop from where I released the stick. There is a difference from teorical distance based only on deceleration and true distance.
A good results near to end point should be the same, move the stick at center (stop) just in time for stop the camera on  the end point. Maybe with a corrector parameter?

Willyfan, 2018-08-14:
About vesc6 I opened a issue about configuration some days ago. Maybe I'm using vesc6 in wrong way. I set it in FOC, How do you set your VESC?

DmitriySurin, 2018-08-14:
I use high values for brake. Also I can very smooth stop my cablecam manually moving slow the stick to neutral. Also the braking distance from what I can during my tests not longer than at operational mode. So the main question for me why under the same condition the result is too different?

DmitriySurin, 2018-08-14:
Also I dont understand how the cable cam can know the speed if the diameter of the halls wheel and the distance between the magnets can be so different from cable cam to cablecam? What is the frequency of calculation the braking distance? One hall step or what?

Willyfan, 2018-08-14:
@DmitriySurin 
The cablecam don't know the absolute speed in m/s, but at pulse/second. As also distance in measured in pulse and not metres, it's ok

wernerdaehn, 2018-08-14:
@Willyfan  Okay, will document my settings on the VESC next weekend.

@DmitriySurin The statement "I can smoothly stop my cablecam manually moving the stick to neutral" is not a fair comparison. The controller does not let the motor slow down with its settings but by the programmed acceleration value. A fair comparison would be you want to stop the motor faster than with drag brake alone, hence you have to put the stick in neutral once a while. This would obviously not be as smooth as the drag brake alone.
I'd like to proof that by you setting a lower acceleration value in the controller. This should then never kick in the emergency brake.
What I wonder about is the slow speeds and slow accel I see in your video. I hope it is not something else. Do you have bluetooth working menawhile? Would be interesting to see the emergency brake output during the drive.

DmitriySurin, 2018-08-14:
I try lower accell settings and have the same result.    Actually I have to spikes during braking. After applaying emergency brake the board immedeatly realease brake to the actual or lower position of stick and let the cablecam moving and then cycles this one more time before reaching end point. The lower accell value the bigger distance between thoose two choppy spikes. But thoose spikes, errors, choppy braking or whatever it calls is still here. I have function called "servo speed" in my radio so I can emulate boards accell values. My bluetooth is working so I send you the log of the braking cycle. What video you mean?

Willyfan, 2018-08-14:
@wernerdaehn Video is mine....

Willyfan, 2018-08-15:
I had some logs about my deceleration.
I remember that I'm using VESC.
Itried with tolerated error 100 and  300, and in left and right direction.
This is the log
```
Tolerated error 100 - to left
Endpoint brake on. Distance= 866.000, calculated braking distance= 917.956
Endpoint brake on. Distance= 848.000, calculated braking distance= 908.956
Endpoint brake on. Distance= 830.000, calculated braking distance= 899.956
Endpoint brake on. Distance= 811.000, calculated braking distance= 940.453
Emergency brake on 
Endpoint brake on. Distance= 789.000, calculated braking distance= 1077.946
Emergency brake on 
Endpoint brake on. Distance= 765.000, calculated braking distance= 1163.941
Emergency brake on 
Endpoint brake on. Distance= 743.000, calculated braking distance= 1055.946
Emergency brake on 
Endpoint brake on. Distance= 721.000, calculated braking distance= 1044.946
Emergency brake on 
Endpoint brake on. Distance= 699.000, calculated braking distance= 1033.946
Emergency brake on 

Tolerated error 100 - to right

Endpoint brake on. Distance= 866.000, calculated braking distance= 911.739
Endpoint brake on. Distance= 848.000, calculated braking distance= 902.739
Endpoint brake on. Distance= 830.000, calculated braking distance= 893.739
Endpoint brake on. Distance= 811.000, calculated braking distance= 933.891
Emergency brake on 

Tolerated error 300 - To left
Endpoint brake on. Distance= 854.000, calculated braking distance= 917.956
Endpoint brake on. Distance= 836.000, calculated braking distance= 908.956
Endpoint brake on. Distance= 817.000, calculated braking distance= 949.953
Emergency brake on 
Endpoint brake on. Distance= 794.000, calculated braking distance= 1138.444
Emergency brake on 
Endpoint brake on. Distance= 771.000, calculated braking distance= 1126.944
Emergency brake on 
Endpoint brake on. Distance= 749.000, calculated braking distance= 1066.946
Emergency brake on 
Endpoint brake on. Distance= 727.000, calculated braking distance= 1055.946
Emergency brake on 
Endpoint brake on. Distance= 705.000, calculated braking distance= 1044.946
Emergency brake on 
Endpoint brake on. Distance= 684.000, calculated braking distance= 986.949
Emergency brake on 
Endpoint brake on. Distance= 634.000, calculated braking distance= 697.463
Emergency brake on 

tolerated error 300 -  to right 
Endpoint brake on. Distance= 861.000, calculated braking distance= 864.998
Endpoint brake on. Distance= 843.000, calculated braking distance= 906.880
Endpoint brake on. Distance= 824.000, calculated braking distance= 947.762
Emergency brake on 
```
What I not understand is that the braking distance aftewr a while seem to raise and not lower, and this is impossible because the cablecam slow the speed.

Willyfan, 2018-08-15:
Analyze for example this log:
```
Endpoint brake on. Distance= 866.000, calculated braking distance= 917.956
Endpoint brake on. Distance= 848.000, calculated braking distance= 908.956
Endpoint brake on. Distance= 830.000, calculated braking distance= 899.956
Endpoint brake on. Distance= 811.000, calculated braking distance= 940.453
Emergency brake on 
Endpoint brake on. Distance= 789.000, calculated braking distance= 1077.946
Emergency brake on `
```
At begin the distance to the endpoint and braking distance decerase, but not how much it need. Anyway, the error is less than 100.
At the fourth cycle, the breaking distance increase suddenly. this is a no sense because to have this result the cablecam must accelerate. but I see that cablecam is in deceleration.
At fifth cycle is increased againa lot, but at this point the brake stop the camera finally.
I don't know what happen, but there is a error in braking distance calculation. I see no other explanation.

wernerdaehn, 2018-08-15:
I agree. In order to narrow it down, I will add more information.

The distance is calculated in those lines:
    float time_to_stop = abs_d(getStick()/controllerstatus.stick_max_accel);
    float distance_to_stop = controllerstatus.speed * time_to_stop / 2.0f;

I suspect that the speed is not accurate enough. For example it might switch between the values 5 and 6 and that might be enough to cause that. I have a second way of measuring the speed, that is by measuring the time between two Hall pulses. This way it is not old_pos-current_pos every 0.02s and therefore never a decimal but a natural number.

Willyfan, 2018-08-15:
Ok. I think that to do a complete debug we need for every cycle also:
Virtual stick value, or the erpm sent to vesc
Real speed. 
The fact is that the deceleration, when I release the stick at zero is very, very smooth, so I'm sure that we can have the same results at the endpoint. 
Please let me know, I can try again from next Monday.
I have also some other ideas (like hyperlapse function), but I will explain later.

Willyfan, 2018-08-16:
How is calculated the value for stick_max_accel?
Because, we input a value from 0.001 to 1, but in your code you transform it with line
`activesettings.stick_max_accel = p[0]*CONTROLLERLOOPTIME_FLOAT;`
What value is CONTROLLERLOOPTIME_FLOAT ? I found it in config.h:
`#define CONTROLLERLOOPTIME_FLOAT 0.02f`
but I don't understand how work ...

Willyfan, 2018-08-16:
Ok, understood. The loop time is 20 ms, so if the input value is 1 (max) the cam decelerate in 1 sec. If the value is 0.5, the cam decelerate in 2 seconds ans so on. Stick can have value from 0 to 1, so you subtract every loop cycle the max_accel value in order to go at 0 in definited time (or, better with a definited ramp, time depends from stick_last_value).
I want check if on my setup this is real. I will put max acc at 0.5, pot at max and try to release the stick, to see if the camera stop in 2 seconds, and how much distance. I can measure, roughly my setup is 175 pulse for meters. Just ti have some more info.

Willyfan, 2018-08-16:
Some test done.
with max speed at around 0.25 and max acc to 0.5, when I release the stick the camera stop in 2 seconds (so, it's correct).
In 2 seconds the camera runs 4metres and 20 cm, around 735 pulse in my config.
unfortunately in my garden my rope is only 10 meters long, so I can't do better test.
I tried aslo at lower speed (lowered with pot). Camera stop in similar time (maybe a bit less but I'm not sure) but runs less on the rope.
In this situations the deceleration is always smooth. Goal is start with deceleration around end point with this value. As I don't know the speed read when I released the stick, I can't do more.

wernerdaehn, 2018-08-19:
Okay, I made a few changes:
1. The more accurate speed measurement using the time between pulses instead of testing in 20ms intervals.
1. The end point brake and the emergency brake print out all details now. Note that these print out something only the first time the brake kicks in. If you have multiple lines, that means that in one cycle it did brake because of the endpoint, in another cycle the speed was low enough to not reach the endpoint, another cycle it had to brake again.
1. The command $d P enables to print out above position messages in every single cycle. Good for debugging. (A temporary setting only, not effected by $w)

Willyfan, 2018-08-21:
Don't work...
I tried at low speed, because with high speed the cam overshoot the end point a lot (some metres) and I have not so much rope in my garden.
I tried at 2 different acceleration value, 1 (max) and around 0,5 (pot at mid).
The cam seems not decelerate, or not in the same way when I put the stick in mid pos.
$d P don't work, or I don't see any different result in debug.
This is my log:
with acceleration at 1:
```
Endpoint brake on. Distance= 321.000, calculated braking distance= abs( -0.958[%] ) / 1.000[%/s] * 666.667[steps/s] / 2 = 325.896
Emergency brake on. Distance= 139.000, calculated braking distance= abs( -0.698[%] ) / 1.000[%/s] * 666.667[steps/s] / 2 = 245.896

```
With acceleration at 0.5:
```
Endpoint brake on. Distance= 767.000, calculated braking distance= abs( 0.965[%] ) / 0.501[%/s] * 800.000[steps/s] / 2 = 779.358
Endpoint brake on. Distance= 712.000, calculated braking distance= abs( 0.965[%] ) / 0.501[%/s] * 800.000[steps/s] / 2 = 779.358
Endpoint brake on. Distance= 643.000, calculated braking distance= abs( 0.965[%] ) / 0.501[%/s] * 666.667[steps/s] / 2 = 649.465
Emergency brake on. Distance= 617.000, calculated braking distance= abs( 0.955[%] ) / 0.501[%/s] * 800.000[steps/s] / 2 = 779.358
Endpoint brake on. Distance= 535.000, calculated braking distance= abs( 0.925[%] ) / 0.501[%/s] * 666.667[steps/s] / 2 = 629.465
Emergency brake on. Distance= 510.000, calculated braking distance= abs( 0.915[%] ) / 0.501[%/s] * 666.667[steps/s] / 2 = 622.798
```
In last case, deceleration is choppy. 
the thing that does not convince me at all is the speed value read: 800 step or 666.667 step in this direction.
In the other direction I have also some other values but always inconsistent:
`Endpoint brake on. Distance= 324.000, calculated braking distance= abs( 0.955[%] ) / 1.000[%/s] * 666.667[steps/s] / 2 = 325.159
Endpoint overshot. Distance= -1.000, calculated braking distance= abs( 0.455[%] ) / 1.000[%/s] * 363.636[steps/s] / 2 = 82.814
`
I don't understand what happen, please let me know if I can do other test in order to understand.

wernerdaehn, 2018-08-21:
Okay. One more question: What is the $a value?

Willyfan, 2018-08-21:
$a value is 1, but I done test with pot at max (so is 1) and pot at mid, so is at 0.5 or so (I read 0,501 in debug, I think is correct value).
If you like, I can disable pot and put a precise value for $a. Please let me know.

wernerdaehn, 2018-08-21:
You are right, it was stupid asking. You said the accel is 1.0 at 100% poti, hence I could have known already. 
I would love to test that with lower speeds. 
Assuming the top speed is 50'000erpm for the VESC, a value of 1 would mean it stops from 50'000 to 0 in 1 second. That is significant. If your actual speed is just 5'000rpm and you drive against the end point it would stop in 1/10th of a second.
That is a deceleration way more than physically possible. Hence I would like to set the baseline to something smaller. In my case I have $a 0.5 0.5 which is quite a lot as well and then the poti is at 50%. That gives an effective acceleration of 4 seconds from 0 to max erpm. More reasonable.

How about the following: You limit the max erpm in the controller, say $e 5000. Hence the UART output will be a speed command of 5000rpm at the most. Something you feel comfortable as top speed given the line of the rope. And then way play with smaller accelerations.

I agree with you that above readings are not what I want to see either. What I don't know is if they are a side effect or the root cause. If you want to stop 10kg of mass from 60km/h to 0 within one second, yes the deceleration will not look nice. Simply because you trip the failsafes constantly. Even if you drive just with 6km/h it will not look nice because you essentially stop within one controller cycle. Hence not much difference in output when printing the debug values in every control cycle or not. 

Knot some towels at the ends of your rope so even if you drive over the end point, there is a cushion of some kind stopping the cablecam?

Willyfan, 2018-08-21:
Tomorrow morning I will try with max ERPM at 5000 (actually is at 35000) but:
1) My test was at not higher speed, but with pot at less than half. I don't know the exact value because the curve for this pot is not linear but expo (programmed in the radio). If I try with high  speed the cam go some metres over endpoint, As I told before.
2) If I release the stick in the mid of rope, the cam stop in 2 seconds (as expeted) with acceleration at around 0,5, without problem and very smoothly. Obviusly, it go more metres as the speed is higher. So, I think that the problem is not the inertia of the camera, it CAN stop at the speed I try.
3) in my garden  I have a 14 metres rope, not more. So, I'm unable to go at very high speed because the cam have not time for accelerate. Anyway, the goal is have the same stop at end points as when I release the stick. 
4) Today, at same speed that I test the end point, in the mid of rope when I release the stick the cam stop in 2 second and in around 2 metres or little more. As my cabelcam have around 175 pulse/metres, I expected the braking distance to be around 350 or 400 pulses, but it seems to be much more from debug info (800 pulse). Probably there is still a problem of calculating the speed. 
Tomorrow I try also to take a video, so you can see the speed of the camera.

wernerdaehn, 2018-08-22:
The 400 pulses expected vs the 800 pulses calculated I can explain, I believe. I have configured the encoder so that one pulse is a rising and falling flank in either of the two Hall sensors. So when one magnets does pass, it is 4 steps, not two. (Encoder Mode X4)

Willyfan, 2018-08-22:
Yes, I know, but my evaluation take account of this. I have 175 pulse / meters read from the board. A magnet pass every around 12mm, so if I count the magnets pass they are half then the 175 value.
I will do some other test with low speed in the morning. I let you know asap. 
Also stick, I use stick at 100%, I reduced max speed with pot. When you say "full speed" you mean absolute max speed (for example 50000 erpm) or  only stick at maximum?

wernerdaehn, 2018-08-22:
Reducing the speed with the dial is fine as well. The logic is 100% stick = max_erpm * dial%.

Willyfan, 2018-08-22:
A important note: in the debug I see the stick value that decreases but non in the right way.
For example, in my last test: when endpoint brake start, we are at 271 pulse distance from the endpoint. Stick is at 0,675 (in play mode is max value, I don't know why) and calculate breaking distance is 272, so is ok.
But when we are at half distance, for example at 130 pulse from endpoint, I expected that stick is half than start, so around 0,340. Instead, it is at 0,594. It is decreased, but not as we need.
Beside, speed calculation have some problems: when endpoint breaking start is at 400.000 step (very strange this number), It decrease at 363,636 but raise again at 400,000 and decrease at 363,636, after that 333.333 and so on. And note that speed decrease only when emergency brake  is activated.

wernerdaehn, 2018-08-22:
Okay, that gives me something to think about.
One suspection I have is around the direction, the $r setting. In have to brake when you are driving towards the endpoint, not away from one.

Can we check that setting one more time?
You execute $p. Note the current position.
Then you put the stick forward and write down the current position again.
And with $i I'd like to know if stick forward is above or below neutral.
Finally the value of $r itself. 

To me it looks as if you are driving into one direction, say from pos=0 to pos=1000. After a while you have a high speed, one that has a brake distance of 200. And now the brake kicks is because you are only 190 steps away from the start position. But that is the wrong comparison, you are 810 steps away from the end position you are driving towards. 

A second test is to manually pull the cablecam over the endpoint. Can you drive it back into the save zone with the stick but not further over the end point? That would indicate that $r is correct.

One thing I will do for sure is add the direction as another information to the end point brake output.
Should I do that right away and you update the firmware?

Willyfan, 2018-08-22:
Videos: stop at 5000 erpm: https://youtu.be/_DdwgY-AjL8
another stop at 5000 erpm: https://youtu.be/rhBIw9xJVUs
stop at 10000 erpm: https://youtu.be/i4BbIEvY5mY

Willyfan, 2018-08-22:
> A second test is to manually pull the cablecam over the endpoint. Can you drive it back into the save zone with the stick but not further over the end point? That would indicate that $r is correct.

Just done this test, is ok, If I put cablecam over the endpoint I'm able to go only in safe direction

Willyfan, 2018-08-22:
Now I do also the other test, But I'm sure that $r is correct, because with $r in other direction the camera don't work at all...
And, sure, i will update firmware and test.
A question: your cablecam don't have this problems? I have a friend that he built a cablecam like yours, but he had same problems at the endpoint....

Willyfan, 2018-08-22:
I have $r set to +1.
I done some other test but I used my app because for some strange reason the terminal raed debug but not command so i don't lost much time.
I put the cablecam on the rope, switcht it on and move it slightly by hand
check position give me -60 1716 -11
I put stick to to the first end point. Cablecam stop and position check give me -60 1716 -63
In this position cablecam go only to the last endpoint (to right in my set). I put the stick for some second and stop it. Check pos give me -60 1716 515
$i give me for channel 1 the value of 1014 (this is strange because all other channel in mid position  give me 992, but I think that this is not the problem).
And $r is set to +1, as i just told at begin of this post.

wernerdaehn, 2018-08-22:
Okay, I suspect I have a problem in the code when $r +1. I can test that here on the bench myself. Will work on it in the next minutes.

DmitriySurin, 2018-08-22:
I have the same problems whitin the regular esc too as discribed above.

wernerdaehn, 2018-08-22:
I can't see the problem on my bench test. Can you please try with the firmware I just uploaded. I have added better output when using $d P.

Willyfan, 2018-08-22:
please give me one hour and I send you a log.

Willyfan, 2018-08-22:
Ok, updated firmware and tried.
the logs (always at erpm max 10000, and in manual mode, not autoplay).
I done some test without gimbal, and some test with additional weight of gimbal.
No gimbal, to left direction
```
Startpoint brake on. pos-brakedistance= -47663.000 - 563.947 = -48226.945 < -48220.000, braking distance= abs( -0.990[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Startpoint brake on. pos-brakedistance= -47736.000 - 563.947 = -48299.945 < -48220.000, braking distance= abs( -0.980[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Startpoint emergency brake on. pos-brakedistance= -47762.000 - 558.233 = -48320.234 < -48220.000, braking distance= abs( -0.970[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 558.233
```
no gimbal to right direction:
```
Endpoint brake on. pos+brakedistance= -47177.000 + 563.947 = -46613.055 > -46625.000, braking distance= abs( 0.990[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Endpoint brake on. pos+brakedistance= -47117.000 + 563.947 = -46553.055 > -46625.000, braking distance= abs( 0.990[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Endpoint emergency brake on. pos+brakedistance= -47081.000 + 558.233 = -46522.766 > -46625.000, braking distance= abs( 0.970[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 558.233
```
with gimbal left direction:
```
Startpoint brake on. pos-brakedistance= -47667.000 - 563.947 = -48230.945 < -48220.000, braking distance= abs( -0.990[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Startpoint emergency brake on. pos-brakedistance= -47823.000 - 501.090 = -48324.090 < -48220.000, braking distance= abs( -0.868[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 501.090
```
with gimbal right direction:
```
Endpoint brake on. pos+brakedistance= -47186.000 + 563.947 = -46622.055 > -46625.000, braking distance= abs( 0.990[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 563.947
Endpoint brake on. pos+brakedistance= -47123.000 + 558.233 = -46564.766 > -46625.000, braking distance= abs( 0.970[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 558.233
Endpoint emergency brake on. pos+brakedistance= -47048.000 + 529.661 = -46518.340 > -46625.000, braking distance= abs( 0.919[%] ) / 0.507[%/s] * 571.429[steps/s] / 2 = 529.661
```
In every test I can see the intervention of the emergency brake (also for the "sound" of motor).
As usual, I also test to release the stick in the mid of rope, and camera stop smoothly in around 2 seconds and maybe 3 metres.

Willyfan, 2018-08-22:
The strange thing is that speed is **always** 571.429 step/s. There is something wrong in this.

wernerdaehn, 2018-08-22:
Please reduce the accel significantly and also $d P so we can see the intermediate values.

While you are driving the output of $p would be interesting as well as I print out the two speeds, based on the interrupt and based on the pos difference. Maybe that gives an indication if the speed is correct or not and which.

It can be that the speed remains the same. As the speed is primarily the time between two hall pulses (rising flank), if there is no hall puls, the value is not updated. And after 2 seconds I assume a speed=0. Can't see why it should remain 571 for more than two cycles but....

Willyfan, 2018-08-22:
I suspect that from bluetooth terminal the command $d P don't work. I try to put command with putty, and I don't switch off the board.

wernerdaehn, 2018-08-22:
Also, this might be too much text for the bluetooth connection...

wernerdaehn, 2018-08-22:
I definitiely don't like the speed reading. Let me revert that back to pos-difference only.

Regarding the "you don't start at zero" we need to keep an eye out. There should not be a difference to before. I certainly don't store the current position anywhere. The only thing that can change the position is when switching $r. Because then the current position is inverted.
I'll upload the revised code in a minute.

Willyfan, 2018-08-22:
Ok.
Now is raining a lot, so i can do test as it finish...
Anyway, as yopu can see, the position is not more from zero and value around it (before I had from 0 to 1700) but is at -650000 or so. I don't know what happen.

IMPORTANT NOTE: as my rope is not so long, the deceleration start before that the camera was at max speed, or, to say better, before that stick arrive to 100%. In both cases, the stick is around 0.650 when endpoint brake start. But there is an inconsistence: when brake start, the calculated brake distance is around 540, in both cases. But, if you see how much distance the camera had traveled (-658426 - (-659609)) results is 1183 pulses. In acceleration, so is not a inertia problem.
What I mean is that with this acceleration and deceleretion setting it need almost the same pulses to stop, and not 540 that is half of this. I don't understand where is the conceptual problem, But I'm pretty sure that is this.

wernerdaehn, 2018-08-22:
So you are saying the distance calculation is wrong? By a factor of two?
Let's go into the math with this line:
braking distance= abs( -0.607[%] ) / 0.266[%/s] * 363.636[steps/s] / 2 = 421.818

The stick is at 0.607 and can change by 0.266 per second. Therefore it takes 0.607/0.266=2.3s for the stick to go to zero.

The actual speed is 363steps/s. The distance covered when braking to zero within 2.3s is 363*2.3/2=414steps. s=v*t/2.

https://www.lernhelfer.de/sites/default/files/lexicon/image/BWS-PHY-0261-07.gif

Willyfan, 2018-08-22:
it is obvious that if the camera reaches the maximum speed and runs a stretch at that speed all this calculation does not work, but we are lucky because in this case the brake engage while the camera is still accelerating.

Willyfan, 2018-08-22:
Sorry, I read a bad value, when brake start the evaluate speed is 444 pulse/sec. Always half of right value... (right for me, please tell me if I'm in wrong way)

wernerdaehn, 2018-08-22:
I would raise two points:
1. The absolute value of the speed does not matter as it is normalized by time and calculated via the pos sensor. What I mean with that is, according to the stick position it shall take 4 seconds to get to a speed of zero. Now let's assume with your pos sensor the speed would be 1000steps/s, with mine it is 500steps/s. Therefore your distance to travel would be 1000.0*4.0/2=2000steps. With mine it is 500.0*4.0/2=1000steps. So either I travel half your speed and then the deceleration would be just half as well. Or we drive equally fast and I jave twice as many hall pulses, hence the deceleration would be the same just the unit of distance is different. Either way, it is not the steps that count, it is solely the time. 
1. Look at the speed readings. The values go up and down, these are odd numbers and worse, assuming we have on printout of the debug message per cycle, I can calculate the speed based on the pos change: pos(n) - pos(n-1) = delta_pos_per_0.02s; Hence (pos(n) - pos(n-1))*50 = speed/sec. This value is approx. twice as high as the shown speed. My conclusion is, the speed measurement via the interrup does not work. Hence I have disabled it and go via the pos changes only.

Willyfan, 2018-08-22:
I agree with point 1. What I mean is that if value of speed is wrong, also time to stop is wrong, because brake start too late.
Anyway, If you get the last 2 position value in my log, you see that difference is 20 pulse, so the speed should be 20*50=1000 pulse/sec, not so different from my calculation before (975). Maybe that use pos change is ok. But please don't remove the debug because I remember a problem before you change speed measurament.

wernerdaehn, 2018-08-22:
??? The last two positions in your log are -659575 and -659588, so 13 steps apart. 
The other thing I noticed during debugging is that the prints happen every 50ms, so just 20 times per second. Hence 13*20=260steps/s and your log says 255 as speed.

I also checked again the two types of speeds and both look okay and within the same range. See for yourself with $p, the last two numbers. They are speed by pos_diff and speed by time_diff.

    Startpoint emergency brake on. pos-brakedistance= -659575.000 - 235.000 = -659810.000 < -659592.000, braking distance= abs( -0.490[%] ) / 0.266[%/s] * 250.000[steps/s] / 2 = 235.000
    Startpoint emergency brake on. pos-brakedistance= -659588.000 - 206.667 = -659794.688 < -659592.000, braking distance= abs( -0.485[%] ) / 0.266[%/s] * 222.222[steps/s] / 2 = 206.667

Even the loop time of 50ms versus the execpted 20ms does not matter as I actually do the calculations per loop. The times 50 (or times 20) is just for viewing.

Willyfan, 2018-08-22:
I mean last two position before brake. Is 20 point different, 659113 and 659093. In that moment speed is around 1000 pulse per second, and it is right...

wernerdaehn, 2018-08-22:
Okay

    pos= -659073 speed=400.000[steps/s]
    pos= -659093 speed=400.000[steps/s]

Within one cycle the delta_pos=20. Therefore the speed=20steps/cycle. Under the assumption that we run 50 cycles per second, the speed would be 1000steps/s, not 400. Just as you said. 

The stick is at -0.644[%] and the acceleration is 0.266[%/s] which means 0.00532%/cycle. Hence it takes 121 cycles to get to zero. Brakedistance = 20steps/cycle * 121cycles / 2 = 1210steps. My calculation shows 480 because it is based on the speed=400.

In the next cycles the stick is

    -0.639[%]  
    -0.634[%]
    -0.628[%]

Assuming that this is per cycle, it is the required reduction by 0.00532%/cycle. So that is correct.

Summary: The speed per pos_diff is wrong. 
Before I calculated the speed simply by the difference between previous and current position. That has the problem that the resolution is just 1step per cycle. What if the speed is so low that it takes two cycles for one step? I wanted to capture that and therefore divide the diff by the number of cycles between pos changes. 
I went though it five times now and cannot spot the issue. Will debug it further.

Thanks for your help, well done!

Willyfan, 2018-08-22:
A stupid idea: use more cycles for speed calculation? Last 2 or 3?
Or,  a better solution, put more magnets on the reel? Or an encoder with more resolution? I just designed a board for a AS5047p, same chip I want use on vesc, Maybe is useful for cablecam?

Willyfan, 2018-08-22:
Another  way, suitable with vesc. During cablecam setup, we move it on the rope at a high speed for some meters. So, we can automatically calculate ratio from  erpm and pulse/meters. We can save this parameter, and use the erpm speed of the motor when speed read from hall sensor fall down under a threshold, for example 15 or 20 pulse/cycle or less. 
As the camera travel at low speed, I think that there are no inertia problem.

Willyfan, 2018-08-23:
Is impossible to read the frequency of input pulse from our hall sensor?

Willyfan, 2018-08-23:
ok

wernerdaehn, 2018-08-23:
I can't find the error because there is none. I just ran a test where I output every single position information and the speed is correct. So yes, please try with the currently available firmware, the one that is using the pos difference only.

Meanwhile I will make code changes to reduce the variability in the speed measurements. The root cause of that is due to the fact we do not run the code at 50Hz but less.

[75 more replies]