wernerdaehn/CC3D-CableCam-Controller #32: Buzzer for end points.

DmitriySurin, 2018-09-11:
Is it possible to use buzzer outputs of the flip board  as the confirmation of setting the endpoints? One beep for the first and two beeps for the second endpoint? Also it be useful to dyplicate onboard mode light via led strip output.

DmitriySurin, 2018-09-21:
Any idea?)

wernerdaehn, 2018-09-22:
Implementation is probably simple. Need to find a spare GPIO pin, PWM output and we are good.
But is it worth it? You would need to attach a buzzer. For this one use case. Often you want to be as quite as possible. and you get the same output if you would use a bluetooth connection plus more.

What would be your arguments offsetting this?

DmitriySurin, 2018-09-22:
The arguments is very strong...Carrying a 30 000 USD camera onboard I want be for 100 sure thats my endpoints are set propertly by any reason. Any operation should have the feedback from the device. Flip board have separate buzzer outs.

wernerdaehn, 2018-09-23:
That's an argument for getting feedback. But not if bluetooth or buzzer, is it? Yes, the pins are not used at the moment.