wernerdaehn/CC3D-CableCam-Controller #34: help for configuration fly sky fs-th9x

mackios, 2018-12-05:
hello,
i m not an exper in rc model and programation.
my configuration
-lrp 1600kv
-skyrc toro 150 amp
-radio fly sky fs-th9x
-recever fs-r9b (8 channel) pwm whis an encoder pwm to sumppm for connecting 1 wire in the cc3d revo
-hall sensor well and capter

all is connected,
it's neccesary to configur the skyrc toro 150 amp before programing the cc3d?
i have flashing the cc3d whis the firmwear, i set this fonction $I0 for sumppm and now i don't no set this comand $j whis my radio for make the end point, max speed...etc
if any can help me 
thank you nicolas

mackios, 2018-12-05:
[image]
[image]
[image]

wernerdaehn, 2018-12-05:
Hardware looks good.
You are you connected with the CC3D you said and can send commands, yes?

Then I would like you to show me the following:

- What is the output of $I command?
- What is the output of the $d p command?
- What is the output of the $i command?

Details here: 

https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/docs/Step%203b%20-%20Receiver%20Input%20SumPPM.md

mackios, 2018-12-05:
ok thank you.i test this tomorow end , take screem shot of the result.

mackios, 2018-12-06:
hello.
when i connect this morning the cc3d to putty... this is not possible to connect...driver lost?no detection whis putty?
the cc3d have a blue and green led fix, no blue led blinck.
is possible to reconnect after flashing a firmwear whis dfuse.
it's normal?

wernerdaehn, 2018-12-06:
The output of $d p I don't like. There is no long pause used as indicator for a packet end, some values are negative - how is this even possible?? - and the values are definitely too small.

So either a problem on my side or the converter does something unexpected. Do you have a link which converter you are using? Maybe I can find something that makes sense.

And a way to see the actual signal you do not have either, an oscilloscope or something similar, I assume.

Another approach is to use a different receiver, one that supports the SBus protocol. I just did some reading and it says the Th9x https://oscarliang.com/flysky-tx-rx-buyers-guide/ is using an older protocol for which there are no SBus/SumPPM receivers. The newer ones do not support SBus either but another similar protocol which might be possible to make compatible with the CC3D. 
Obviously I would prefer SBus.

mackios, 2018-12-06:
the converter is this https://fr.aliexpress.com/item/8CH-Receiver-PWM-to-PPM-SBUS-DBUS-S-BUS-32bit-Encoder-Signal-Converter-FUTABA/32817294343.html?spm=a2g0s.9042311.0.0.40696c3747XXcS .
no i look my radio th9x ....and it is not a pwm protocol...it's ppm or pcm
i have an others radio, futaba t8j whis sbus, i use this for control the alexmos gimbal whis sbus.
tomorow i take this for test, if you say sbus is better i take this for my cablecam and i test control the gimbal whis th9x.
thank you for your help

wernerdaehn, 2018-12-06:
The documentation in you link says:

> When one of CH1-CH8or some of CH1-CH8 input PWM signal, CH9 will output PPM signal and CH10 output SBUS signal. (LED Indicatorflash 1 second)

So in your case, since you have connected PWM signals, the LED indicator should flash with 1Hz. And since you have connected Pin10 with the CC3D, it should be configured as SBus. That would also explain why the $d p shows so low values.
Can you simple change $I 1 and then show $i output again?

mackios, 2018-12-07:
hello
comande $i avec modification $I1 and i connect in the converter on the sbus
[image]

mackios, 2018-12-07:
and $dp
[image]

mackios, 2018-12-07:
normaly, if i move the stick of the radio and i make comand $i the result of this comand $i maybe change?

mackios, 2018-12-07:
> normaly, if i move the stick of the radio and i make comand $i the result of this comand $i maybe change?

it's good when i change the value of the stick radio the value in the comand $i change.
whis th9x in sbus.
after i don't no if the value for $dp is good?

wernerdaehn, 2018-12-07:
I am skiiing. But the $i output looks good. Yes, move and hold the stick in
an extreme position and $i will show a different value for one channel.

mackios, 2018-12-07:
ok.thank
i test the comand $1 and set the $j.

> hello.
> when i connect this morning the cc3d to putty... this is not possible to connect...driver lost?no detection whis putty?
> the cc3d have a blue and green led fix, no blue led blinck.
> is possible to reconnect after flashing a firmwear whis dfuse.
> it's normal?

;)
and for this it's normal

mackios, 2018-12-07:
I buy a tanaris , a same of you . It’s better for your help me😀

mackios, 2018-12-17:
hello
i configure the tanaris, i think same of your.
look the $i
[image]
and look when i pute the diferent switch
[image]
it's good?
and when i make this comand $1, i have a problem look the picture.
[image]
and in the pictures i don't have this fonction
[image]

thank for your help

wernerdaehn, 2018-12-18:
Yes, that all looks good. Very good in fact.
It seems in the Taranis a mixer is programmed or the stick assigned to multiple output channels.
https://oscarliang.com/taranis-mixer-for-wings/

On the weekend I can download my Taranis settings and you can import them. Let's use that as Plan B.

mackios, 2018-12-18:
ok thank.
my tanaris is an helicopter mode , it's good? or plane mode is better?
for sitting all stick and switch it's in input or mixer?
MY FIRMWEAR TANARIS IS  opentx-x9d+ vers 2.2.2 is good?

wernerdaehn, 2018-12-18:
You can start with heli-mode, which is using a lot of mixers. And then you disable the mixers. It does not matter which mode you start with as default setting. 
Plane mode usually has less mixers but I personally would not start from scratch again.
Firmware version does not matter.

mackios, 2018-12-18:
ok.
this link 
https://oscarliang.com/taranis-mixer-for-wings/
it's for set my radio?

wernerdaehn, 2018-12-18:
Yes, this is how a mixer might look like and how you can enable/disable it. But there are other resources like the manuals.
You goal is: In the Servo-Display page, the one you showed in you last screen, whenever you move a stick or flip a switch, one and only one channel should change. At the moment it seems two are changing when you move the stick from full forward to full reverse.

mackios, 2018-12-18:
i have only one channel moving for the esc, i do not understand why it's not happening.i am going to try to just activate that the esc to see if it's working

wernerdaehn, 2018-12-18:
In that case I see two options
a) Try again using the $1 command but make sure you move the stick just up/down and not left/right. I have some tolerance built in but maybe you did not pay attention closely enough.
b) Set the channels and the ranges manually. The $1 command exists to help but is not the sole method.