wernerdaehn/CC3D-CableCam-Controller #36: CC3D mini revolution no input signal

lockheedm, 2018-12-16:
Hi,
I'm using cc3d mini revolution from hobbyking I have spektrum dx8 and I'm using ppm encoder.
I can't get any input signal from receiver.
with $i command all values are zero. I have tried with another 3ch radio with no result. I also changed to another ppm encoder and still no input signal. I'm starting to think that this cc3d board is not compatible with this firmware.
Please help me. 
Connected RC input according to this picture:

[https://cdn-global-hk.hobbyking.com/media/file/441019279X769333X13.jpg](url)

wernerdaehn, 2018-12-18:
Overall, I expect the same result - it will not have enough or the same pins. But the PPM input we should manage.
On the CC3D Revo the Receiver Input is on the Pin PA10, which is a pin that serves UART3. Assuming the two boards are not totally different, that would be the pin called FlexiBoard RX. 
Is this where you have connected the receiver to?

wernerdaehn, 2018-12-18:
In that case the output of $d p would be interesting. Can you post it here to see if it has any resemblance with a PPM signal.

lockheedm, 2018-12-18:
Here is the new situation.
Reflashed firmware, connected same as before printed $i values and they change without me moving any channels .
Here is $d p values

[image]

lockheedm, 2018-12-18:
And one new problem..
After flashing firmware I press leave dfu mode on dfuse, and board boots normally (blue led flashes) and commands in putty work. But if I dissconect the board from usb and then reconnect it green and blue led's are solid and board is unusable. 
Becomes usable again if I re-flash it and till next usb reconnection.
Seems like this board simply won't work in cable cam application.

wernerdaehn, 2018-12-18:
The $d p output shows a normal 8 channel PPM signal. So $i should look okay and stable. Does it?
The other problem you describe is odd. This would happen if there is an uncaught issue that happens only if you boot it the one way and not the other. No immediate idea.

lockheedm, 2019-02-08:
could not make mini revo to work, so i bought flip 32 f4.
Got it working but run into a small problem, is there a command to restore all settings to default?

wernerdaehn, 2019-02-08:
Yes, $W.
$w writes the settings, $W does erase the eeprom.