wernerdaehn/CC3D-CableCam-Controller #37: eeprom save failed

lucaspezzei, 2019-02-12:
Hello.

I want to build this project, but if i type $w in putty to save the settings, putty says "eeprom save failed". what should i do? i use this board:

https://www.banggood.com/it/Original-Airbot-Omnibus-F4-V5-Flight-Controller-OSD-STM32-F405-MCU-MPU6000-5V-1A-BEC-p-1319176.html?gmcCountry=IT&currency=EUR&createTmp=1&utm_source=googleshopping&utm_medium=cpc_bgs&utm_content=frank&utm_campaign=usc-ods-it-css&creative=312667001418&gclid=Cj0KCQiA-onjBRDSARIsAEZXcKYIIycO5e6uoVse5AGQ9Wk1ZeAxUo0rSKFHlM3M5TMi9SM_oixUc8EaAod5EALw_wcB&cur_warehouse=CN

thank you.

wernerdaehn, 2019-02-13:
I am testing if the connected chip is an EEPROM during init. For that I check for specific vendor IDs and my guess would be on that board it is a different one. 
So three options
1. I remove the check
1. You can tell me what EEPROM chip is used
1. We read the EEPROM vendor ID and I add that to the list of expected ids

But with the Omnibus I have the doubt that all pins are available. Have you checked some already?

wernerdaehn, 2019-02-13:
Correction, I do not check for the vendor id. That part got removed already. So it is the write itself that has a problem. Hence I would need to know the used EEPROM.

lucaspezzei, 2019-02-13:
Thank you for the fast reply. Yes, the omnibus chip has all the pins needed. For the EEPROM, i`m not so expert in the microchip world, there is a picture showing the model number on the chip, hope it help to solve the issue.

[image]

[image]