wernerdaehn/CC3D-CableCam-Controller #43: End point won't work

litrosy, 2019-10-17:
I finally set everything together and put it on the rope. Everything works fine but... Something strange is happening with the endpoint function. After entering the program mode , I set end point 1 and endpoint 2, but when I return to operational mode, ccam does not respect the set endpoint settings. Like I didn't even set end points.

My $r setting is r-1 (as I drive forward, the pulse rate decreases), but i also tried it with $r1, (I get a message:" Is $r really correct? Does not look like it") but with no luck.

 ESC is brushed 
Motor is brushed

litrosy, 2019-10-17:
This are the data obtained through the bluetooth module:
 First test with $r1:

18:10:17.769 Connected
18:10:17.809 K
18:10:23.191 $p
18:10:23.229 $p^M$p -276 1121 118.000 0.000 0.000 OK*4f
18:10:23.243 
18:10:34.030 $p
18:10:34.448 $p^M$p -276 1121 671.000 0.000 0.000 OK*47
18:10:34.548 
18:10:42.418 $r
18:10:42.864 $r^M$r 1 OK*47
18:10:42.867 
18:10:52.019 $p
18:10:52.340 $p^M$p -276 1121 671.000 0.000 0.000 OK*47
18:10:52.383 
18:11:02.268 Entered Endpint Programming mode
18:11:14.238 Point 1 set
18:11:42.171 Point 2 set
18:11:42.175 Is $r really correct? Does not look like it.
18:11:42.176 Drove from 186.000 to 2444.000
18:11:43.165 EEPROM Write successful
18:11:52.142 Entered OPERATIONAL mode
.............................Nothing happens..............

Test with $r-1:

18:12:36.462 Connected
18:12:36.500 K
18:12:44.528 $r
18:12:44.567 $r^M$r 1 OK*47
18:12:44.573 
18:12:56.104 $r-1
18:12:56.467 $r-1^M$r -1 OK*6a
18:12:56.509 
18:13:08.470 $p
18:13:08.934 $p^M$p 186 2444 -46.000 0.000 0.000 OK*7c
18:13:08.946 
18:13:28.842 Entered Endpoint Programming mode
18:13:31.333 Point 1 set
18:14:02.256 Point^VWH($r really correct? Does not look like it.
18:14:02.265 Drove from -46.000 to -2438.000
18:14:03.253 EEPROM Write successful
18:14:13.228 Entered OPERATIONAL mode
18:14:34.446 passthrough with li[W
18:14:37.446 passthrough with limiter & end points
.............................Nothing happens..............

Test with $r0:

18:15:19.725 Connected
18:15:19.764 K
18:15:27.129 $r0
18:15:27.169 $r0^M$r 0 OK*46
18:15:27.170 
18:15:34.127 Entered Enpoint Programming mode
18:15:41.610 Entered OPERATIONAL �ode
18:15:43.109 Entered En����PɅ�����mode
18:15:53.081 Pont  set
18:16:11.534 Point 2 set
18:16:12.033 ESC direction not set yet. Correct value seems to be $r 1
18:16:12.534 Saving all settings ($w) to store the $r parameter permanently
18:16:13.035 Drove from 2763.000 to 469.000
18:16:13.037 EEPROM Write successful
18:16:20.383 $w
18:16:21.882 $w^M$w
18:16:21.917 Settings saved successfully OK*1d
18:16:21.918 
18:16:24.420 Entered OPERATIONAL mode
[... 126 lines cut ...]
18:17:14.419 $p^M$p t69 2763 820.000`0.000 0.000 OK*6b
18:17:14.435 
18:17:24.256 $r1
18:17:24.395 $r1^M$r 1 OK*47
18:17:24.428 
18:17:28.636 $w
18:17:30.009 $w^M$w
18:17:30.014 Settings saved successfully OK*1d
18:17:30.014 
18:17:35.398 $p
18:17:35.866 $p^M$p 469 2763 820.000 0.000 0.000 OK*6b
18:17:35.901 
18:17:37.530 Entered Endpoint Programming mode
18:17:41.809 Point 1 set
18:17:52.781 Point 2 set^]
18:17:53.284 Is $r really correst? Does not look like it.
18:17:56.775 Drove from 820.000 to 2344.000
18:18:04.254 EEPOM Write stccessful
18:18:15.224 Entered OPERATIONAL mode
18:19:00.449 passthrough with limiter
18:19:07.830 passthrough
18:19:27.445 passthrough wb���ѕ�
18:19:48.451 Disconnected

.............................Nothing happens..............

litrosy, 2019-10-17:
The only thing common to all attempts is the message:
"Is $r really correct? Does not look like it."
And I notice there are a lot of spelling errors and symbol appearances when printing a report, but this is probably a bug of the mobile app?

is this normal

wernerdaehn, 2019-10-18:
What did you find out? I wanted to suggest to turn on the debug messages for driving...

litrosy, 2019-10-18:
Solution for driving between endpoints were in $o settings. It was "none" by default. I try from 0 to 1000 and $o300 seems to be fine for my setup.

As far as typing errors are concerned, they still exist and I do not know how to deal with it. I will try with debugging... 

And... I still get the warning message ("Is $r really correct? Does not look like it") but everything is working normally.

closed 2019-10-18 by litrosy