wernerdaehn/CC3D-CableCam-Controller #46: Increased range of bluetooth module

litrosy, 2019-11-29:
Hi @wernerdaehn ,
I'm wondering if there is any other bluetooth module option that would have a range greater than 100m or some other two-way communication solution. It would be useful to have a range at least equal to the largest rope reel, 200m.

What do you think?

wernerdaehn, 2019-11-30:
The normal CC3D Revo has a OPLink 733MHz long range wireless module. I wanted to utilize that as well but it is not that straight forward, hence never started writing that code module.

litrosy, 2019-12-02:
Um, yeah. That would really be something. I mean, as it is now, it's fine. Is there any chance of getting caught up in that? What range are we talking about, do you think it would be 200m?

wernerdaehn, 2019-12-02:
https://librepilot.atlassian.net/wiki/spaces/LPDOC/pages/33620037/OpenPilot+OPLink+Mini

> With it's peek power output of 100mW, using good quality antennas, useful range spans over 2km.

litrosy, 2019-12-02:
That would be something! Too bad that its integration is a big problem. With such a range, it would certainly cover the many scenarios and needs that may arise in the field. Of course, this is not bad either. It's great to have such a long range with bluetooth.  If you ever get some free time... That would be huge. Really great work and research so far!

closed 2019-12-08 by litrosy