wernerdaehn/CC3D-CableCam-Controller #47: PWM frequency

litrosy, 2020-03-02:
Hi @wernerdaehn ,
I changed a new ESC that can operate at 48V. When I connect everything as it was on the previous 24V system (and everything worked fine), nothing happens. The ESC does not respond at all. 

The ESC manual says:
 "-PWM mode1: 0-100% PWM: You can use your own PWM controll signal from your PLC or
microprocessor. The minimum controll PWM frequency is 100Hz. The step resolution 1us. The
acceptable voltage level: 3.3-10V.

- PWM mode2: RC wirelles receiver input 1.5msec pulse for 0% duty cycle and 2.5msec pulse for
the 97% duty cycle."

Currently is $n 1500 20 800, but whichever else I put, it is not responding at all.  I was thinking that the problem is maybe in PWM frequency which is below the required minimum of 100Hz? 
In "wired" Potentiometer or throttle mode, everithing is fine.

Any suggestions?

wernerdaehn, 2020-03-02:
I have seen non-responsive ESCs in those cases where the signal **at start** was not within the neutral range of the ESC. The mode2 text above says "1.5ms to 2.5ms". Wouldn't that indicate that the neutral position is 2ms? So $n 2000 20 480? I hope that works. It is a bit unusal.
https://github.com/wernerdaehn/CC3D-CableCam-Controller/blob/master/docs/Step%205a%20-%20Regular%20ESC.md

The mode1 sounds like a true phase-width-modulation. Which means a signal that consists of x ms ON and y ms OFF. x+y=constant and 10ms minimum. Either way, we do not support that. We support RC like pulse modulation, where every n ms a puls is sent with the length of p ms, where p is something between 0.5 and 2.0ms. 
https://en.wikipedia.org/wiki/Servo_control

litrosy, 2020-03-02:
Thank you very much for your quick response. 

The problem was exactly what you wrote. I don't know how I missed $ n2000 20 480. 

Thanks for your time, you really do help a lot.

closed 2020-03-02 by litrosy