wernerdaehn/CC3D-CableCam-Controller #52: Control board

Freeejack, 2021-05-24:
Hi,

I see that the project is stopped two years ago, neverthless I'am very interested to  want to build your cablecam, but i am stuck at the start, cannot find any of the mentioned CC3D controller on the market.
Is it possible to change it to another type, or i need to find another solution and this project is EOL?

Thanks!

wernerdaehn, 2021-05-24:
Define "stopped" ;-)

I am still looking at it, the code is fine and I agree the CC3D controllers used cannot be found in the market. There are however many other CC3D controllers which all should work. The pins might be differently placed but as long as there is a circuit diagram for the PCB, the documentation files in this repo explain which MCU pin is used for what. Plan B is to measure which Pin goes to what MCU pin.

Do you have a CC3D PCB in mind? Please provide a link.
Have you found the schematics for it?
Would you suggest to upgrade the code to something different?

Freeejack, 2021-05-24:
Sorry for the word, I just thought I didn’t see a new update. But then it was not updated because the system is working properly. ;-)

I'am not very good at programming, rather mechanical parts, therefore my plans are not complicated. I want to control an oDrive controller and motor (from Your project links), and use the endpoints functionality for safety.
I have few Graupner receivers and a transmitter.

So if i good understand only CC3D controllers good for the job, and other controllers cannot be used?

(Or is there any "simpler" method to control the motor control board, for example direct from the receiver? (Sorry if this a dumb question))

wernerdaehn, 2021-05-24:
The odrive allows positional control, so you can program the start and end points there. No need for this controller.
There are however two issues:

1. How to program the start and endpoints? Must be done via USB/UART/Bluetooth whereas the cable cam controller does that for you using switches on the RC sender.
2. What if there is slippage on the motor wheel? And there is some, always. Not much with low accelerations and level rope, a lot when driving up/downhill often.

You got the point that a CC3D is not so much a single PCB but a schematic and many companies manufacture PCBs in a compatible manner? If you pick any, chances are good it will work. But as said, the precise documentation would help.

Graupner, I have to admit, I do expect issues. Their SumPPS signal is different to all others. I prefer SBus receivers.

Freeejack, 2021-05-24:
Ok, and thank for the help.

I found one used board, i hope this can be used.
I Here is the manual of the board: [manual](http://tricopter.hu/docs/cc3d_manual.pdf)

Thanks again, and sorry for the misunderstandings!

reffff21, 2022-02-01:
> Hi,
>
> I see that the project is stopped two years ago, neverthless I'am very interested to want to build your cablecam, but i am stuck at the start, cannot find any of the mentioned CC3D controller [...]

No problem to find it in Aliexpress
https://aliexpress.ru/item/32800037994.html?spm=a2g2w.productlist.0.0.1cc325cd3NhWFE&sku_id=64212712917

Willyfan, 2024-06-05:
About the board: I'm searching some board with a STM32v4, I found the Holybro Kakute F4 V2.4 or SpeedyBee F405 V4. Last one have bluetooth onboard. I'm not able to find schematics diagram, but it seems to be usable for this project (maybe with only little adjustement). What do you think? Someone know this boards?