wernerdaehn/CC3D-CableCam-Controller #56: Roller Wheel

Unyck, 2022-04-26:
I have a simple question, how do you manage to drill the in line wheels properly ?  The polyurethane is slippy i try to attach it in a vice below a drill press but the materiel escape the wick. 
If you have any advice about that

wernerdaehn, 2022-04-26:
I have created a fixed axis as holder and then drilled with the CNC mill/lathe the holes.

Unyck, 2022-04-30:
I read on one of your YouTube channel than you also use hard material wheels, I try on really soft one, it might be why I struggle.
Do you talk somewhere what is the size of the rope you use, I think about 8 or 12mm
Also I like to know the material of the spacer do you use nylon or aluminum ?
I can ask somewhere else if needed, glad you respond to me so fast, I really appreciate all of your work

closed 2022-06-16 by Unyck