wernerdaehn/CC3D-CableCam-Controller #60: Bluetooth  4.0 le problem

Willyfan, 2024-09-24:
Dear Werner, 
As I have a lot of problem with recent android with bluetooth 2.0, so I tried to connect some 4.0 le board.
I set board from pc by terminal, and connection from phone and pc using a usb serial adapter is working.
But if I connect the board to cc3d it don't comunicate.
I done a lot of experiments, without luck.
Can you help me about this? Maybe sothing must be change in the source code, or maybe you know a board that work.
Many thanks

wernerdaehn, 2024-09-25:
The most obvious one would be different baud rates or uart settings. Didn't I implement code that tries different things to get the BT Uart connected?

Willyfan, 2024-09-25:
the best bluetooth board I have is a DSD-TECH HM-10 (https://www.deshide.com/product-details.html?pid=344835&_t=1665209850). They have also a their tools for setting (it's a serial tool) I set the board for correct bitrate (38400) and I'm sure that it's working at that bitrate because if I connect it to a My PC by a USB serial adapter, I can transmit and receive from a terminal on my phone.
Maybe UART settings need to be changed, but is a field that i don't know very well, So I ask for your help.
I'm using your firmware, I haven't tools for build on my PC but i can work on it.
Beside, the DfuSe utility don't work anymore but on ST site they have a programmer thet work well with HEX file, but this is another problem.

Willyfan, 2024-09-26:
Hi Werner, after a lot of try and settings, the DSD-TECH is working. But i must rewrote the app for Bluetooth LE. Thank you anyway!